Guizz27/Marlin---Longer-LKx-Pro #7: DGUS file update for Marlin 2.0.9.3

boromyr, 2022-01-26:
Hi, I adapted your modification to dgus reloaded to work with the latest version of Marlin. It should work fine for all pages, the only problem is the print pause which I think doesn't work well, plus the offset z during printing, but that's not Marlin's problem
[dgus_reloaded.zip](https://github.com/Guizz27/Marlin---Longer-LKx-Pro/files/7942444/dgus_reloaded.zip)
.

Guizz27, 2022-01-27:
Thank you very much! I have integrated Marlin 2.0.9.3 on my local copy but I haven't had time to test it and refresh the github repository. I will compare what you did with my version. I may catch issues on my version before I test it.

Guizz27, 2022-01-28:
I have an action in Photoshop called "saveall" that will activate layers and save for each screen. Before running it, only the background layers should be visible. I recorded that action once by exporting each screen one by one.

boromyr, 2022-02-03:
Could you share it? I've never used Photoshop, but I think I understand it's a script or macro. I'd appreciate it.

Guizz27, 2022-02-05:
Here is the file.
[LK4Pro.zip](https://github.com/Guizz27/Marlin---Longer-LKx-Pro/files/8008002/LK4Pro.zip)

boromyr, 2022-02-07:
Thank you very much

boromyr, 2022-02-08:
@Guizz27 I saw that you updated the reposity files, did you make any corrections or changes? In that case I will update my files as well.

Guizz27, 2022-02-11:
It's the Marlin 2.0.9.3 integration including some of your changes. There is more than the lcd code to integrate for pause and powerloss recovery for example. My repository is still a few commits behind Marlin. I will do another integration soon. 
There are no new features or fixes. I'm working on a tool to help for the screens. Once I have this tool, I will add more features.

boromyr, 2022-02-11:
Ok thanks, so for now I continue to use my files until your next updates.

Sawurai80, 2022-02-18:
buonasera, ho compilato marlin 2.0.9.3 per la mia lk4 pro no bltouch e mi sono accorto che la regolazione dello Z-off durante la stampa da due problemi: se si regola con le frecce si verifica lo spostamento di z ma non viene incrementato il suo valore sullo schermo, se invece si vuole inserire il valore attraverso il pannello di input, quest'ultimo si apre ma in maniera incompleta...potresti gentilmente aiutarmi a risolvere il problema?

Sawurai80, 2022-02-18:
potresti gentilmente linkarmi il file firmware dello schermo? te ne sarei veramente grato.
il mio problema è questo:
[image]

Sawurai80, 2022-02-18:
potrebbe dipendere dalla compilazione errata di marlin?

boromyr, 2022-02-18:
No Marlin has nothing to do with it, the screen has its MCU and its memory, the graphics are managed internally. The screen source is [here](https://github.com/Guizz27/DWIN_LK4Pro), but you need DGUS 7.3 to open it.

Sawurai80, 2022-02-18:
cosa conviene che io faccia? grazie ancora per la tua disponibilità

Sawurai80, 2022-02-19:
risolto il problema della schermata incompleta ... ho riformattato la SD e flahato il DWIN SET dentro la cartella release... ma rimane il problema che qualsiasi valore inserisco su adjustament z-offset il valore sulla schermata rimane sempre 0.

Guizz27, 2022-02-19:
This control sets the Z-offset. If modified while printing it will adjust it using baby stepping.
Since you don't have a BLTouch, the value displayed is always 0. Also, anything you enter is ignored.

In a future release I could make baby stepping available to non-BLTouch users.

boromyr, 2022-02-19:
Did you solve the problem by simply reloading the firmware again?

Sawurai80, 2022-02-20:
ciao ragazzi ... ho formattato lentamente la SD in FAT32 e 4096 byte, poi ho inserito la cartella DWIN da release e infine ricaricato il firmware sul display. così la schermata di input numpad si è aperta correttamente.

boromyr, 2022-02-20:
Great thanks then I'll try too!

Sawurai80, 2022-02-20:
> Questo controllo imposta l'offset Z. Se modificato durante la stampa, lo regolerà usando il baby stepping. Dal momento che non hai un BLTouch, il valore visualizzato è sempre 0. Inoltre, tutto ciò che inserisci viene ignorato.
> 
_> In una versione futura potrei rendere disponibile il baby stepping agli utenti non BLTouch._

il babystepping per gli utenti non BLTouch sarebbe veramente una grande cosa!!!! potresti gentilmente lavorarci su??
grazie in anticipo.

boromyr, 2022-02-21:
On thingvierse there is an adapter to mount endstop Z instead of bltouch and emulate it, you could try this alternative. Theoretically, you can enable bltouch in marlin by leaving endstop Z where it is, you will be able to homing and babysteps, but you will not be able to use ABL.

Guizz27, 2022-02-21:
> il babystepping per gli utenti non BLTouch sarebbe veramente una grande cosa!!!! potresti gentilmente lavorarci su?? grazie in anticipo.

I will work on it but I'm busy and won't be able to work on it for a little while.

Sawurai80, 2022-02-21:
> On thingvierse there is an adapter to mount endstop Z instead of bltouch and emulate it, you could try this alternative. Theoretically, you can enable bltouch in marlin by leaving endstop Z where it is, you will be able to homing and babysteps, but you will not be able to use ABL.

gentilmente potresti indicarmi qual'è l'adattatore e come procedere per fare questo? grazie mille

boromyr, 2022-02-21:
I don't know why I didn't save it, you should do a keyword search, like: "bltouch endstop z replace". The main image I think was an Ender 3 extruder with this white support. The instructions on how to proceed I believe are in the description.

Sawurai80, 2022-02-21:
[image]
È per caso questo?

boromyr, 2022-02-21:
Exactly, with this you can probe the bed, and you can also use the screen firmware for bltouch

Sawurai80, 2022-02-21:
grazie mille di tutto! vi aggiornerò sui risultati

jgbrown54, 2022-04-11:
When compliing for an LK5 without BLTouch, I get an error saying DGUS requires a bed probe. Is there a version for manual leveling?

Guizz27, 2022-04-12:
Sorry, after integrating 2.0.9.3 the compilation without BLtouch was broken. I just committed a fix. The change is in the file src/inc/SanityCheck.h. I tested that it compiles correctly, I didn't test if the code works on the printer since the code is the same as the one I did a while ago to support non-BLtouch printers.

Guizz27, 2022-04-13:
Yes it works for both. Without a BLtouch, when you select Leveling in the main menu it goes directly to the 5-point manual leveling screen. The user will not see options for auto leveling and Z-offset.

Without a BLTouch, the Print Adjust screen still has a Z-offset option at the bottom you would normally use to adjust during the first layer, this option is there but does nothing for non-bltouch. In a future version, I will make sure it works.

closed 2023-06-03 by Guizz27