Guizz27/Marlin---Longer-LKx-Pro #11: Display does not work after changing board

Roman7603, 2022-04-28:
Hello, I am trying to change the board on LK4 pro (BTT SKR 2.0b) I used your firmware modification with no problems on 
the LGT_KIT_V1.0. I compiled the firmware for the new board without any big problems and everything works, even the display. After starting the printer I can only see the Marlin logo on the display. The display cable is the same as for the previous board.  (I also tried to change RX TX for display) Display still works on previous board. On the new board, the only difference is on the TFT port. The order of pins is the same, but the new board has 5th pin RST.

My question:
Do you have some experiences with LK4pro display and an other board than LGT_KIT_V1.0?
Is there some exact binding for the display and LK4pro stock board?
Do you have any idea or tips how to solve this problem?

additional info:
I changed the board, because this 32bit board can give me more options that I need -  two separate Z axis, UART mode, next PWM cooler

I think that It would be a good idea to have a solution on how to change the board without changing the display, because your modification is perfect!
Thank you for your work and time.

Guizz27, 2022-04-29:
I don't have any experience with different boards. 
Maybe you have to change the serial port? 
in configuration_adv.h 
#define LCD_SERIAL_PORT 1 

Maybe it needs a different value.

Roman7603, 2022-04-29:
Good news...
I found exact binding for LGT board and DGUS display - Serial 1 is originally reserved to DGUS LCD
After changing LCD_SERIAL_PORT to 1 and SERIAL_PORT to another value than 1 the  display works with new board:-)
Maybe It will help someone  who want to upgrade board.

I will test more,  pre release configuration is in my fork of your repository.

closed 2022-04-29 by Roman7603

Guizz27, 2022-04-29:
This is nice! Thanks! Make sure USB still works.