Guizz27/Marlin---Longer-LKx-Pro #12: Enable RUNOUT_DISTANCE_MM 25 in Configuration.h

jgbrown54, 2022-06-07:
An ongoing issue with Longer is false Runout detection. Marlin has enabled this option to allow a longer period before action is taken. It basically says that when a Runout is detected, don't take action unless 25mm of filament has been extruded. If the condition is cleared before the 25mm is extruded, no action will be taken and the counter will be cleared. This will prevent noise caused by longer wires to the sensor and momentary detections caused by filament diameter errors of less than 25mm. Actual filament runout will still occur. It will just be slightly delayed.

Guizz27, 2022-06-08:
Currently integrating Marlin 2.0.9.4 and 2.1. I will enable this be default.

Guizz27, 2022-06-27:
This is turned on in the new release 2.0.9.4 GC 1.1.

closed 2022-06-27 by Guizz27