Guizz27/Marlin---Longer-LKx-Pro #14: Set default NOZZLE_TO_PROBE OFFSET to 50mm

jgbrown54, 2022-07-01:
#define NOZZLE_TO_PROBE_OFFSET \
    {                          \
        -50, -6, -1.2          \
    } // Value for Alphawise and Longer U30 pro LK4 pro  with BLtouch support => https://www.thingiverse.com/thing:4261004

// Most probes should stay away from the edges of the bed, but
// with NOZZLE_AS_PROBE this can be negative for a wider probing area.
#define PROBING_MARGIN 50

I recomend the above changes due to the Probe Offset when configured with DUAL Fans from Longer being 50mm.

Without the second change, the probe will sometimes miss the left edge of the bed resulting in bad leveling. It will also cause the probed area to be non uniform because the nozzle is prevented from going past the left edge of the leaving the probe 50mm from the edge. With the probing margin set at 50, a uniform matrix is probed allowing for better leveling performance.

The probe offset can be set by entering an M851 gcode. But the probing margin can not be set by the user.

jgbrown54, 2022-07-02:
Probing_Margin should not be less than the NOZZLE_TO_PROBE_OFFSET. This can cause 2 issues when using ABL. 1. The probe is at an X of 0 and may miss the edge of the bed. 2. When probing to the right, the Nozzle stops at the MAX_X of the bed leaving the probe at MAX_X - Probe_Offset. This results in a non-uniform probing matrix. Probing Margins section in Configuration_adv.h should account for the probe offset.

Guizz27, 2022-07-11:
I didn't want to set the NOZZLE_TO_PROBE_OFFSET to 50 since this does not match stock printer LK4 Pro and older LK5 Pro I think. So I left it to -32 which can be set with M851.

I also did not change the default PROBING_MARGIN.

Instead, I added a new setting called EQUAL_MARGIN_ALL_SIDES in Configuration.h that is enabled by default.
When enabled, it will automatically use a margin equal to the largest X/Y probe offset. It also works with M851. So you don't have to change the configuration.h file, you can set your offset -50, -6, -1.2 with M851 and the margin will automatically adjust to 50.

closed 2022-07-11 by Guizz27