Guizz27/Marlin---Longer-LKx-Pro #17: Longer config.h?

eugenejanvrin, 2023-01-15:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

jgbrown54, 2023-01-15:
The following lines in the Configuration.h file are all that need to be changed to select the type of printer.

// @section machine
#define LGT_MAC //  For Alphawise and Longer U30 pro, LK4/LK5 pro
#define LK4_Pro //  for LK4pro, U30 Pro
//#define LK5_Pro //  for LK5pro
#define LKPro_BLTOUCH // for LK4Pro and LK5Pro

#define LKPro_FW_VERSION "GC 1.2"
#define LCD_HEIGHT 4 // number of message lines in the wait screen
//#define DEBUG_DGUSLCD
//#define FILAMENT_RUNOUT_SENSOR_DEBUG

// Choose the name from boards.h that matches your setup
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_LONGER3D_LKx_PRO //  For Alphawise and Longer U30 pro LK4 pro with LGT motherboard
#endif

eugenejanvrin, 2023-01-15:
wow im an idiot didnt realize the default configuration.h in the zip file was the one i needed thank you very much

eugenejanvrin, 2023-01-15:
i guess im doing it wrong because after uploading the firmware i cant move any steppers or preheat.. i also see no other menu options

Guizz27, 2023-01-15:
Have you also updated the screen firmware?
You also have to make sure the sd card used to update the screen firmware is formatted according to the specs (Max 32GB, formatted FAT32 with 4096KB clusters).
Instructions at the bottom of this [page](https://github.com/Guizz27/Marlin---Longer-LKx-Pro)

eugenejanvrin, 2023-01-15:
I have not updated the screen yet but only because I'm not getting any temp readouts and I cant manually move any axis'.

Guizz27, 2023-01-15:
Are you trying from the screen or from usb like with gcode or software because it will not work with the original screen firmware, it is not compatible.

Also, the firmware requires homing the axis before they can be moved.

Temp readout should still work from a usb connection although I never tested it without a screen firmware update.

eugenejanvrin, 2023-01-15:
I will test the screen firmware now

eugenejanvrin, 2023-01-15:
Well unfortunately I cannot format any of my sd cards to 4096 bytes :(

Guizz27, 2023-01-15:
Try to open a Command Prompt (on Windows hit the windows key, then type "cmd") and enter the following command. 
`FORMAT D: /FS:FAT32 /Q /A:4096`
 Replace D: with the drive letter assigned to your SD card.

eugenejanvrin, 2023-01-15:
OK

eugenejanvrin, 2023-01-15:
Invalid parameter.. all my sd card are 32 gb.. i may have a smaller one ill look

eugenejanvrin, 2023-01-15:
i miss typed something on the first try 
this is the message i get

C:\Users\Eugene>FORMAT G: /FS:FAT32 /Q /A:4096
Insert new disk for drive G:
and press ENTER when ready...
The type of the file system is FAT32.
QuickFormatting 28.8 GB
The specified cluster size is too small for FAT32.
Format failed.

eugenejanvrin, 2023-01-15:
Victory!.. I used win10 disk management tool to delete partition of my sd card the added a new one but only 8 gb the was able to format at 4096 bytes.. every thing works fine.. i love the new look and setting up the bl touch will be much easier.. 

thank you both for your help

Guizz27, 2023-01-15:
You beat me. I was trying to write instructions on how to re-partition. Glad you figured it out and you could do it. Also glad that it now works!

eugenejanvrin, 2023-01-15:
its still thinks its an LK4 with a 220x220x300 build though.. i think i can figure that out though

Guizz27, 2023-01-15:
Make sure you comment out LK4 and uncomment LK5 in Configuration.h
//#define LK4_Pro //  for LK4pro, U30 Pro
#define LK5_Pro //  for LK5pro

eugenejanvrin, 2023-01-15:
im not sure what im ding wrong but when i comment the lk4 setting out and uncomment the lk5 settings in the build fails.. i know im doing something simple wrong but at his point im too tired to think.. im going to get some sleep.. please please feel free to comment with some suggestions on what im doing wrong.. im fairly new to this

jgbrown54, 2023-01-15:
Why don't you just download one of the pre-compiled version from here: https://github.com/Guizz27/Marlin---Longer-LKx-Pro/releases/tag/2.1-GC-1.2

eugenejanvrin, 2023-01-15:
jgbrown54 thank you very much.. i simply did not know it existed.. im pretty new to this and teaching myself as i go.. that worked took all of 45 seconds thank you again

eugenejanvrin, 2023-01-18:
[Guizz27] do you know anything about adding dual z steppers to the firmware

Guizz27, 2023-01-18:
I'm sorry, I don't know exactly the steps to enable and calibrate dual Z steppers.  But let me know if you find the information and need help to recompile the firmware.

eugenejanvrin, 2023-01-18:
OK I will, i need to figure out the "pinout" for the Z2_STEPPER_MOTOR

Guizz27, 2023-01-18:
According to an old pinout diagram it seems the second Z is simply in parallel with the first one since there seems to be only one stepper for both Z. 

 
[LGT_KIT_V1.0_004 PIN.pdf](https://github.com/Guizz27/Marlin---Longer-LKx-Pro/files/10441822/LGT_KIT_V1.0_004.PIN.pdf)

eugenejanvrin, 2023-01-18:
Ive been trying to find picture again i found it but it was blurry.. seems i should just be able to copy the pin readout

eugenejanvrin, 2023-01-18:
not an easy task so far

Guizz27, 2023-01-18:
for the second stepper motor, you just connect it to the second Z connector on the board. Nothing else to do.
I don't know the steps to calibrate both axis (make sure they are at the same height) when both stepper motors are in parallel (use the same stepper driver).

eugenejanvrin, 2023-01-18:
i had done that but nothing was happening. turns out i needed to swap the two middle wires on the connector

eugenejanvrin, 2023-01-27:
can i use UBL with this firmware

Guizz27, 2023-01-28:
I never tested it. There are no controls on the screen for UBL but I think it could be controlled with gcode commands using a remote host.
You can turn it on in the code and test it if you want.

bob291, 2023-06-03:
Does this have the firmware for the BL touch? I installed a CR Touch, and now I'm not sure what to do. I designed and printed a bracket and re-pinned the connections. But I don't know what to do about the firmware. I have updated the Motherboard firmware to 0.4.0-Marlin2.x and the screen to 0.3.1-DW.

Guizz27, 2023-06-03:
Yes there is a version for BLTouch in the releases page but this is not the official Longer software. If you use this firmware, you also have to upgrade your screen firmware (also in the releases page). Be sure you have a screen that is compatible if you choose to use this one. this firmware has more features than Longer's official one. Make sure you have a DW type screen (in your about screen).

Longer also has a BLTouch version of their firmware but the link seems broken. You may have to contact them.

macheen, 2023-06-12:
I have an incompatible screen the one with a j after it. is there any way i can reconfigure this to work for me or where can i get a DW screen?

jgbrown54, 2023-06-12:
You can get a screen from China on aliexpress [https://www.aliexpress.us/item/3256804359976899.html?](url). Make sure you get the touch screen. Or you can run without the display with something like the Pixie [Pixie-Server.com](url) and move your control to a browser.