Guizz27/Marlin---Longer-LKx-Pro #18: lounger lk5 pro dual blower fan.

ffarae, 2023-01-29:
i have the lounger lk5 pro with the screen modal 0.3.2-DWTJB, am trying to get the Bltouch to measure the whole bed and also be able to heat the nozzle up to 300C or higher can you please help I been trying for weeks to edit my own marlin code but it doesn't work for me. am still a noob at this kind of stuff. i want to be able to control the temperature from the screen also but the screen it self is limited to 250C.

Guizz27, 2023-01-30:
Are you trying to edit Longer`s Marlin or this one?
This Marlin is not compatible with the screen 0.3.2-DWTJB. If you want to keep your screen you will have to use the official Longer`s Marlin code. If you are willing to replace your screen you can get the DWT48270C043_06WT (same model I have). For example on [aliexpress ](https://www.aliexpress.com/item/1005002234042314.html?spm=a2g0o.productlist.main.3.4d8f743aKwRyqF&algo_pvid=3cdd746e-76c8-4085-94dc-4e963e1847a0&algo_exp_id=3cdd746e-76c8-4085-94dc-4e963e1847a0-1&pdp_ext_f=%7B%22sku_id%22%3A%2212000019456635096%22%7D&pdp_npi=2%40dis%21CAD%2116.09%2116.09%21%21%21%21%21%40211bd7d616750414415658275d06d9%2112000019456635096%21sea&curPageLogUid=l6mlxcNWplsI) 

If you keep your screen and Longer`s firmware, you will have to find which software to use to change the temperature limit on the screen. Also, without proper knowledge, you won't be able to get auto leveling on the screen. As far as the temperature limit in Marlin I think it should be fairly easy to modify. I'm guessing you have a proper hotend for that kind of temperature.

ffarae, 2023-01-30:
Thank you for your response, i do have a all metal hot end it's just i
wanted more screen features. So the software you have basically have
control over temperatures that's higher then stock and all the other great
futures if that the case i will get the screen you mentioned. Again thanks
alot for your time.

jgbrown54, 2023-01-30:
If you order one from DWIN, it comes without the white connector for the cable from the motherboard. You'll have to add it. If you order one from Longer, you'll get the wrong display. It also takes 30 days or more to get one from China. If you are in the US, I have one left I can sell you for $35 including shipping and you can get it in about 4 days.

Guizz27, 2023-01-30:
This firmware has no temperature limit on the screen but has a limit of 275 if I remember in the firmware, so you will still have to change it in the code and re-compile it.

ffarae, 2023-01-30:
Thanks for letting me know. I think am going to hold off for now, i have a
big project coming and I don't want to get stuck with the process upgrading
the printer specially for something major like this, but it's nice now i
have that option later on down the line. I wish there was a YouTube channel
that actually do this kind of thing, every YouTube channel out there they
just review it and call it a day.

eugenejanvrin, 2023-02-15:
The screen upgrade is well worth it and if you have everything ready before hand then the downtime on the machine is very minimal

closed 2023-06-03 by Guizz27