Guizz27/Marlin---Longer-LKx-Pro #21: Implementation of Marlin 2.1.2.1

jgbrown54, 2023-06-29:
Attached is a modified DGUS_RELOADED, a new DWIN_SET, and a description of the changes made. You should consider adding these changes. For now they only work for the LK4 Pro and LK5 Pro with BLTouch. I refrained from making changes in Marlin's core firmware and limited my changes to DGUS_RELOADED and DWIN_SET to make it easier to integrate future Marlin upgrades.
[Marlin 2.1.2.1 For Longer.zip](https://github.com/Guizz27/Marlin---Longer-LKx-Pro/files/11902962/Marlin.2.1.2.1.For.Longer.zip)

Guizz27, 2023-07-10:
Thanks but I already integrated Marlin 2.1.2 on my computer and I do not plan to use the DWIN project anymore. I wrote a program to generate the DWIN_SET folder from text files. It makes it easier to add new features without having to open photoshop and DGUS software.
I'm still testing the changes and trying to add a few more features. I will still look at your changes and compare.

jgbrown54, 2023-07-10:
I did miss a few things. So you can trash what I sent you. Do you want an update when I get it working?

Guizz27, 2023-07-11:
I'm fine, I need to fix a couple small display issues.

jgbrown54, 2023-07-11:
close

closed 2023-07-11 by jgbrown54