Guizz27/Marlin---Longer-LKx-Pro #23: SD card not recognized after update

Martin123400, 2023-09-19:
**Describe the bug**
After updating Marlin(with hex file) and the display, the SD card with all my prints wont show up on my Printer. Everything else seems to work perfectly. Thanks for your great work!

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

Guizz27, 2023-09-20:
Were you using official Longer firmware before?
Do you use an extension cable for the SD card?
Do you have filenames with non-english characters?

Martin123400, 2023-09-20:
Hello and thanks for replying! :)
1. I was using the official firmware, but recently I updated the firmware to the offical BlTouch version (I believe it's called Marlin 0.3.1 BL). But this version had a bug that the z Axis would report back a wrong value after homing. So I switched to Marlin 2.1 with your display update - and the z Axis problem is gone.
2. I do not have an extension cable, but when I got the printer I desolderd the original micro SD card reader and solderd the non micro version of it in. Never had problems with that so far.
3. No, all "standard" characters.

Martin123400, 2023-09-21:
Update: I connected the printer over usb to my pc and started pronterface. And funnily enough the command M21 shows whats on the Printers sd card. I could theoretically even start a print from the sd card this way. So that means the problem must be with the display firmware? Just to be sure I updated it another time, but the Problem remains. Any help with that kind Sir? :-)

Guizz27, 2023-09-21:
Thanks a lot for this test and the information about the "No media" message.
The difference between M21 and the firmware is the firmware uses the SD card detect pin. Normally this is a pin on the SD socket that detects the presence of a card. Seems like you don't have that pin or it is not connected.

If you can compile the firmware from sources, you can disabled the test for the SD card detect pin.
In Configuration_adv.h you have to uncomment the line #define NO_SD_DETECT.

Let me know if you prefer I give you a hex file with that change.

Martin123400, 2023-09-21:
If you would give me the hex file that would be much appreciated, because I am too dumb to find the "No_sd_detect" line in the configuration xD Thanks so much for helping me, your work is amazing and I cant wait to test all the new functions :)

Guizz27, 2023-09-22:
You can try this one
[LK5Pro_BLTouch_NoSDDetect.zip](https://github.com/Guizz27/Marlin---Longer-LKx-Pro/files/12695893/LK5Pro_BLTouch_NoSDDetect.zip)

Can't upload the hex file directly. You will have to unzip it.

Martin123400, 2023-09-22:
YES. Works perfectly. Thank you so very much :-)

closed 2023-09-22 by Martin123400

FernandoVoigt, 2023-11-11:
> Thanks a lot for this test and the information about the "No media" message. The difference between M21 and the firmware is the firmware uses the SD card detect pin. Normally this is a pin on the SD [...]

Thanks man. I was with the same problem. Using LK4 Pro