Guizz27/Marlin---Longer-LKx-Pro #24: Support for new Mainboard: MKS Robin e3 v1.1

selektionsrest, 2024-01-09:
**Is your feature request related to a problem? Please describe.**

I can't get this firmware to work on my new LK5 pro with the new mainboard

**Describe the solution you'd like**

Since from what I heard all newly shipped lk5 pro come with this board from factory it would be nice to have support for it.

**Additional context**

I think they don't use the standard pinout of this board since the lcd is connected to the "neopixel" & "bltouch" labeled pins on the board

Guizz27, 2024-01-17:
It is probably easy to support this board but I'm guessing the screen is also a new variant that is currently not supported by this firmware. Could you check the screen firmware version in the about screen? must be a number with no letters or a number with letters DW.

cheat45681, 2025-05-16:
Guizz27 do you have stock screen firmware for the stock 4.3 inch longer screen? using lgt_kit_1.0 board.

cheat45681, 2025-05-16:
> **Is your feature request related to a problem? Please describe.**
>
> I can't get this firmware to work on my new LK5 pro with the new mainboard
>
> **Describe the solution you'd like**
>
> Since from what [...]

I have gotten firmware to work on this board using a st link programmer.

Guizz27, 2025-05-19:
You can find stock firmware on Longer's website.

It is more difficult to support versions of the screen other than DW since different software are needed and for some screen the documentation is almost non-existant for normal person like me.