Guizz27/Marlin---Longer-LKx-Pro #25: runout sensor

loroff, 2024-02-29:
bonjours je viens d installer votre firmware sans probleme .Je travaille avec octoprint et un capteur de filament celui ci  mettait mon imprimante en pause  en cas de detection .maintenant il retract le filament se qui me gene un peu .Sur votre ecran il y a la fonction  "runout sensor" je ne comprend pas le fonctionnement car que je l active ou pas je ne vois pas de difference. merci pour votre reponse

Guizz27, 2024-02-29:
Vous avez fait une sauvegarde dans le EEPROM après le PID? Normalement, la température devrait légèrement dépasser la tempérture cible avant de redescendre.

C'est un message de octoprint? Est-ce que l'imprimante doit être redémarrer?

Si le message apparaît sur l'imprimante, il faut peut-être ajuster les configurations
  #define THERMAL_PROTECTION_PERIOD 40        // Seconds
  #define THERMAL_PROTECTION_HYSTERESIS 4     // Degrees Celsius

Une fois la température cible atteinte, il attend 40 secondes et la température ne devrait pas dévier de plus de 4 degrées. Selon le graph.

Si c'est Octoprint qui affiche le message alors il faudrait voir les plugins. Il semble que Octopod entre autre surveille la température et affiche une notification.

Guizz27, 2024-02-29:
Vous pouvez aussi essayer les valeurs PID originales de Longer
  #define DEFAULT_Kp 22.2
  #define DEFAULT_Ki 1.08
  #define DEFAULT_Kd 114
ou M301 P22.2 I1.08 D114

loroff, 2024-02-29:
bonjours je passe par repetier host pour modifier dans l eeprom je quitte et revient verifier si les valeur son bien rester de meme que via l ecran de mon imprimante les valeur on bien etait pris en compte 
octoprint me notifie que le firmware a envoyer un message d erreur et doit redemarrer ,de meme un message d erreur apparait sur l écran de l imprimante 
pour une cible de 210 degres la temperature arrive a 209 puis redescend vers 200 avant de remonté le messages apparait lorsque je suis proche de 200 degres (je possede egalement une tete avec 2 ventilateur de filament )

Guizz27, 2024-02-29:
je ferai des tests cette fin de semaine

loroff, 2024-02-29:
[image]
peu etre une piste effectivement ici .les valeur de mon firmeware d avant marlin2.0 par contre je ne sais pas du tout comment faire pour rentrer dans l imprimante

loroff, 2024-03-09:
bonjours je precise que ma lk4 pro possede cette tete double ventilateur .peu etre le probleme viendrait de la le refroidissement de la tete  est peu etre 
[image]

loroff, 2024-03-12:
[octoprint (2).log](https://github.com/Guizz27/Marlin---Longer-LKx-Pro/files/14573727/octoprint.2.log)
voici un log ou le defaut apparait

Guizz27, 2024-03-12:
Est-ce que le problème survient pendant l'impression? Normalement, ces ventilateurs ne tournent pas avant le début de l'impression et généralement pas avant la deuxième couche.

Effectivement dans le log, il ne semble pas être capable de chauffer la buse assez rapidement.

Si c'est pendant l'impression, tu peux essayer de faire le PID avec les ventilateurs en marche.

Aussi, le log montre que c'est bel et bien l'imprimante qui fait l'erreur. Je pensais que Octoprint aurait pu générer l'erreur.

Aussi, il faut noter que le "P" de PID est surtout utilisé lorsque que la température est loin de la cible tandis que le "I" sert surtout quand la température est proche de la cible.

Dans ce cas la température est à environ 188C donc 12C de la cible. Peut-être que le mix de P et I fait en sorte qu'il ne chauffe pas assez vite. Par exemple, si tu attends que la température est à 150C avant de mettre la cible à 200C il va probablement chauffer plus vite.

C'est quand même intéressant. Je vais faire le test avec des température semblables à ton log.

Guizz27, 2024-03-12:
Je viens de faire un test. J'ai chauffé à 200C. Ensuite j'ai laissé refroidir vers 185-190C et j'ai remis une cible à 200C. Ça pris pas plus de 15 secondes avant d'atteindre le 200C.
C'est quoi les veleurs de ton PID?

loroff, 2024-03-13:
> Est-ce que le problème survient pendant l’impression ? Normalement, ces ventilateurs ne tournent pas avant le début de l’impression et généralement pas avant la deuxième couche.
bonjours non le probleme ne survient jamais pendant l impression  seulement au moment de la chauffe ou lorsque j enchaine plusieurs impression

loroff, 2024-03-13:
bonjours 
-le probleme ne survient jamais pendant l impression seulement  lors de la chauffe ou quand j enchaine plusieurs impression (l impression terminé les chauffes se coupent et moi je remet en chauffe pour l impression suivante).
-au moment de la chauffe seul le ventilateur de hotend tourne les deux autre sont eteint.
-oui j ai essayé avec repetier -host et le probleme est toujours present.
-les valeur de pid actuelles sont :  P 14.62
                                                      I 0.75
                                                      D 71.59

loroff, 2024-03-13:
je viens de faire le meme test chauffe a 200 degres j ai laisser refroidir a 185 puis remis la chauffe ,resultat le message d ereur est apparut au bout de 19 secondes et mon ma buse n etait qua 184 degres

loroff, 2024-03-13:
jai fait le meme test chauffe a 200 puis refroidissement jusqu a 180-185 le message d erreur est apparut au bout de 19 secondes et je n etait  qu a187 degres

loroff, 2024-03-13:
[image]
je vais peu etre dire une betise mais cela peu venir de la ?

loroff, 2024-03-13:
[image]
je vais peu etre dire une betise le probleme ne viendrait pas d une valeur de "watch_temp_period "trop faible ?

loroff, 2024-03-13:
Désolé je ne sais pas comment on fait

Guizz27, 2024-03-14:
Je peux fournir la dernière version que j'ai et qui n'est pas encore sur github.
C'est LK4 ou LK5 et avec ou sans bl touch?

loroff, 2024-03-15:
il  me semblerai avoir resolut le probleme en essayant votre methode : faire le pid en mettant en marche egalement les ventilateurs de buse .je n ai pas reussit a reproduire le defaut

loroff, 2024-03-15:
pour ma part il s agit d une lk4 pro sans bltouch

Guizz27, 2024-03-15:
Si ca fonctionne pour l'instant c'est tant mieux.
Pour ma version que j'ai à la maison je sais qu'il y a un problème je crois quand on fait "pause" sur l'écran et ensuite on "Abort". C'est pour l'instant le seul problème que j'ai remarqué.

Mais c'est mieux de garder votre version si le problème ne se produit plus.

loroff, 2024-03-16:
ok merci pour cette excellent travail j attend votre prochaine version .

loroff, 2024-03-18:
bon j ai parlé un peu trop vite ,toujours quelques messages d erreur au moment ou je remet la chauffe entre 2 impressions l une derriere l autre je pense que je vais essayer de passer  le "WATCH_TEMP_PERIOD " d une valeur de 20 secondes a 40 secondes.
pouvez vous me dire si les ligne  95 a 109 sont correctement commenté ou decommenté pour une lk4 pro sans bltouch merci.
[image]

Guizz27, 2024-03-19:
oui les lignes sont correctes

loroff, 2024-03-29:
petit retour pour signaler que la modif de "WATCH_TEMP_PERIOD" a fonctionné  vous pouvez fermer le sujet merci

Guizz27, 2024-03-31:
Merci

closed 2024-03-31 by Guizz27