Guizz27/Marlin---Longer-LKx-Pro #27: Why i can't set manually the nozzle temperature from display? it isn't a problem of touch

XABXABEER, 2024-10-23:
Hello Thank you for creating this nice firmware, I have an lk5pro and I installed 2 mosfet 1 for the bed and one for the nozzle if I try to set the temperature display of the nozzle I do not time out error and does not heat blocking the printer and forcing me to restart ,  while if I launch a print from sd or not any problem , also I would like to ask you if I can install the same firmware on alfawise u 30 I ask this because I had many difficulties to set for the firmware for lk5 pro I am not practical and I did everything by myself following the comments on the side of the command lines the problem that I have described is caused by a wrong setting that I did or I have mistaken something else? I would not want to return to the original firmware because your is beautiful and complete thank you for your help if you find time to reply thanks

Guizz27, 2024-10-26:
Hi, If you modified the printer you may want to do a PID test.
The heater timeout may have to be adjusted. I believe the one in this repository has it set to 20 seconds. Nest release I set it to 40 seconds.

Longer LK4 Pro and Alfawise U30 Pro are supposed to be the same printer. So yes you can install it, but you have to use the LK4 Pro version.
Do you download the .hex file or do you compile the code yourself?

XABXABEER, 2024-10-28:
Hello Guizz27
Thanks for the answer , i have a longer lk5 pro ,for the firmware I
downloaded the editable file with platformio or arduino ide because I had
to set the correct quotas for the bltouch and I also wanted to increase the
temperature limit of the extruder up to 265à , I did the pid of the
extruder and bed just installed the firmware then set the values. on your
github channel I saw that the file longerlk5 pro with bl touch is available
if the problem persists try to install the firmware from the beginning both
on the printer and display , your firmware is really complete and very
functional I have certainly made some mess I , in any case in which line
the waiting time for extruder can be changed so I try to increase it again
thank you for your work and for your time .
hello

Guizz27, 2024-10-29:
Try to change the value WATCH_TEMP_PERIOD to 40 seconds. It is at line 289 in Configuration_adv.h

XABXABEER, 2024-10-29:
Thank you Guizz
just a clarification the file LK5 PRO with bl touch not editable with
arduino ide or platformio , but the source file that should be modified to
install it on lk5 pro if I install it as is without changing anything
except the shares of bltouch on alfawise u30 pro could give me problems?
Thank you for your time and patience

Guizz27, 2024-11-01:
Use LK4 Pro .hex file or set LK4 Pro in the configuration file if you want to compile.
I never tested U30 Pro. But it's supposed to be the same printer as LK4 Pro. Make sure you have a copy of your U30 Pro firmware on hand in case there would be a problem.

XABXABEER, 2024-11-01:
Thank you Guizz for your answer

 I think that I try to compile the file I would like to learn and then
practicing in this way it will be easier and is definitely more fun
I’ll let you know how it goes thanks
Ciao